EdgePi-Cloud/edgepi-python-sdk #287: ADC

ikohOsensa, 2023-03-16:
https://github.com/EdgePi-Cloud/edgepi-python-sdk/blob/985fa0f13b387a3b5b636628e051d589d7925541/src/edgepi/adc/README.md?plain=1#L10

https://github.com/EdgePi-Cloud/edgepi-python-sdk/blob/985fa0f13b387a3b5b636628e051d589d7925541/src/edgepi/adc/README.md?plain=1#L29

ADC

ikohOsensa, 2023-03-16:
[image]

Error from running the example code

ikohOsensa, 2023-03-16:
[image]
New Error

farzadpanahi, 2023-03-17:
@sjpark608 isn't this a typo?
```
 from edgepi.dac.edgepi_adc import EdgePiADC 
```
shouldn't it be:
```
 from edgepi.adc.edgepi_adc import EdgePiADC 
```

sjpark608, 2023-03-17:
it is a typo