EdgePi-Cloud/edgepi-python-sdk #298: ADC fails to init

farzadpanahi, 2023-04-29:
```
File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 233, in __init__
   self.__config(checksum_mode=CheckMode.CHECK_BYTE_CRC)
File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 933, in __config
   self.__validate_updates(updated_reg_values)
File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 951, in __validate_updates
   raise ADCRegisterUpdateError(
edgepi.adc.edgepi_adc.ADCRegisterUpdateError: Register failed to update: addx=0x2, expected: 0x0, observed: 0x2
```

this is happening on `farzadedge.local`

sjpark608, 2023-05-01:
> ```
> File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 233, in __init__
>    self.__config(checksum_mode=CheckMode.CHECK_BYTE_CRC)
> File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 933, in __config
>    self.__validate_updates(updated_reg_values)
> File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 951, in __validate_updates
>    raise ADCRegisterUpdateError(
> edgepi.adc.edgepi_adc.ADCRegisterUpdateError: Register failed to update: addx=0x2, expected: 0x0, observed: 0x2
> ```
> 
> this is happening on `farzadedge.local` using `edgepi-python-sdk==1.1.22`

This issue only happens on the farzadedge unit. On newer version of hardware, I wasn't able to recreate the issue.

sjpark608, 2023-05-01:
Swap CM4 and SD card from Integtest unit
```
integtest@integTest:~/projects/edgepi_samples $ source venv/bin/activate
(venv) integtest@integTest:~/projects/edgepi_samples $ pip list
Package           Version
----------------- -------
bitstring         4.0.1
edgepi-python-sdk 1.1.22
install           1.3.5
pip               20.3.4
pkg-resources     0.0.0
protobuf          3.20.3
python-periphery  2.3.0
setuptools        44.1.1
(venv) integtest@integTest:~/projects/edgepi_samples $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from edgepi.adc.edgep_adc import EdgePiADC
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'edgepi.adc.edgep_adc'
>>> from edgepi.adc.edgepi_adc import EdgePiADC
>>> edge_adc = EdgePiADC()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 225, in __init__
    eeprom_data  = eeprom.get_edgepi_reserved_data()
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 177, in get_edgepi_reserved_data
    self.eeprom_layout.ParseFromString(self.__read_edgepi_reserved_memory())
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 138, in __read_edgepi_reserved_memory
    mem_size = self.__allocated_memory(EdgePiMemoryInfo.USED_SPACE.value)
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 81, in __allocated_memory
    check_crc(read_data[:-1], read_data[-1])
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/utilities/crc_8_atm.py", line 328, in check_crc
    raise CRCCheckError(
edgepi.utilities.crc_8_atm.CRCCheckError: CRC check failed with codes: computed=0xf8, expected=0xbc
>>> 
```
- fails the crc when instantiating. Now I wonder how it was passing  CRC 

```
integtest@integTest:~/projects/edgepi_samples $ ls
farzadedge_eeprom  TBRevC#14_Proto  TBRevC#16_Proto  venv
integtest@integTest:~/projects/edgepi_samples $ source venv/bin/activate
(venv) integtest@integTest:~/projects/edgepi_samples $ pip list
Package           Version
----------------- -------
bitstring         4.0.1
edgepi-python-sdk 1.1.22
install           1.3.5
pip               20.3.4
pkg-resources     0.0.0
protobuf          3.20.3
python-periphery  2.3.0
setuptools        44.1.1
(venv) integtest@integTest:~/projects/edgepi_samples $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from edgepi.adc.edgep_adc import EdgePiADC
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'edgepi.adc.edgep_adc'
>>> from edgepi.adc.edgepi_adc import EdgePiADC
>>> edge_adc = EdgePiADC()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 225, in __init__
    eeprom_data  = eeprom.get_edgepi_reserved_data()
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 177, in get_edgepi_reserved_data
    self.eeprom_layout.ParseFromString(self.__read_edgepi_reserved_memory())
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 138, in __read_edgepi_reserved_memory
    mem_size = self.__allocated_memory(EdgePiMemoryInfo.USED_SPACE.value)
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/calibration/edgepi_eeprom.py", line 81, in __allocated_memory
    check_crc(read_data[:-1], read_data[-1])
  File "/home/integtest/projects/edgepi_samples/venv/lib/python3.9/site-packages/edgepi/utilities/crc_8_atm.py", line 328, in check_crc
    raise CRCCheckError(
edgepi.utilities.crc_8_atm.CRCCheckError: CRC check failed with codes: computed=0xf8, expected=0xbc
>>> from edgepi.calibration.eeprom_constants import MessageFieldNumber
>>> from edgepi.calibration.edgepi_eeprom import EdgePiEEPROM
>>> from edgepi.calibration.protobuf_mapping import EdgePiEEPROMData
>>> edge_eeprom = EdgePiEEPROM
>>> with open("farzadedge_eeprom","rb") as f:
...     protobuf_serialized = f.read()
... 
>>> edge_eeprom.EdgePiEEPROM()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'EdgePiEEPROM' has no attribute 'EdgePiEEPROM'
>>> edge_eeprom=EdgePiEEPROM()
>>> edge_eeprom.eeprom_layout.ParseFromString(protobuf_serialized)
7174
>>> eeprom_data_to_write = EdgePiEEPROMData(edge_eeprom.eeprom_layout)
>>> edge_eeprom.set_edgepi_reserved_data(eeprom_data_to_write,MessageFieldNumber.ALL)
>>> edge_adc = EdgePiADC()
>>> stat = edge_adc.get_state()
>>> stat.
stat.adc_1          stat.adc_2          stat.checksum_mode  stat.filter_mode    stat.rtd_on         stat.status_byte    
>>> stat.adc_1
ADCReadFields(conversion_mode=value: continuous, code: ConvMode.CONTINUOUS, data_rate=value: 20, code: ADC1DataRate.SPS_20, mux_p=value: 0, code: ADCChannel.AIN0, mux_n=value: 1, code: ADCChannel.AIN1)

```

- I generated the current eeprom protobuf serialized file using test script
- re-write the eeprom using the SDK module
- ADC instantiated without problem
- having an issue to connect to farzadedge using ssh

farzadpanahi, 2023-05-01:
> > ```
> > File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 233, in __init__
> >    self.__config(checksum_mode=CheckMode.CHECK_BYTE_CRC)
> > File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 933, in __config
> >    self.__validate_updates(updated_reg_values)
> > File "/home/farzadedge/repos/edgepi-device-manager/venv/lib/python3.9/site-packages/edgepi/adc/edgepi_adc.py", line 951, in __validate_updates
> >    raise ADCRegisterUpdateError(
> > edgepi.adc.edgepi_adc.ADCRegisterUpdateError: Register failed to update: addx=0x2, expected: 0x0, observed: 0x2
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > this is happening on `farzadedge.local` using `edgepi-python-sdk==1.1.22`
> 
> This issue only happens on the farzadedge unit. On newer version of hardware, I wasn't able to recreate the issue. Just by looking at the stack trace, it looks like the CRC register didn't get written during the instantiation.

`farzadedge.local`  has the latest hardware.

sjpark608, 2023-05-02:
I'm closing this issue, since we found the issue was purely hardware related where SPI pins were not soldered

closed 2023-05-02 by sjpark608