EdgePi-Cloud/edgepi-python-sdk #316: re-investigate the file system behaviour of PWM character device

sjpark608, 2023-05-15:


sjpark608, 2023-05-16:
There's no need to call close().
- in above scenario, it only requires calling disable() then enable() again to get the PWM working again

sjpark608, 2023-05-16:
To avoid avoid the above edge case, I propose the following solution
1. set_config method (pwm_num, frequency, duty_cycle, polarity)
   - only called for the first time it is being initiated
   - handles configuring specified pwm
   - checks whether the PWM is already enabled, if it is, disable it.

farzadpanahi, 2023-05-16:
> To avoid avoid the above edge case, I propose the following solution
>
> 1. set_config method (pwm_num, frequency, duty_cycle, polarity)
>
>    * only called for the first time it is being initiated [...]

@sjpark608 I think this is a good idea. user needs to call enable after set_config, right?
also I think you need a check in enable function to see if it is already enabled do not enable it. is that right?

sjpark608, 2023-05-16:
@farzadpanahi  @jarell-santella 

Re-tested opening the same PWM device in separate processes.
1. As long as it is not accessing the file at the same time, a separate process can access the same device
2. if one process closes the device, the device is also closed on the other process

farzadpanahi, 2023-05-16:
@sjpark608 what happens if they try to access the file at the same time?

sjpark608, 2023-06-16:
I guess this issue is resolved with what we found during device manager and collector troubleshooting

closed 2023-06-16 by sjpark608