EdgePi-Cloud/edgepi-python-sdk #332: PWM cannot have 2 instances at the same time

jarell-santella, 2023-05-25:
If I start up 2 terminals running the following 2 scripts:

Script 1:
```py
import time
from edgepi.pwm.edgepi_pwm import EdgePiPWM
from edgepi.pwm.pwm_constants import PWMPins
 
start_time = time.time()
 
try:
    while True:
        pwm = EdgePiPWM()
        pwm.init_pwm(PWMPins.PWM1)
        print(f"Frequency reading: {pwm.get_frequency(PWMPins.PWM1)}")
        time.sleep(1)
except Exception as exc:
    print(exc)
    print(f"Failed in {time.time() - start_time} seconds")
```

Script 2:
```
import time
from edgepi.pwm.edgepi_pwm import EdgePiPWM
from edgepi.pwm.pwm_constants import PWMPins
 
FREQUENCIES = [1234, 3456, 5678, 7890]
 
counter = 0
 
start_time = time.time()
 
try:
    while True:
        pwm = EdgePiPWM()
        pwm.init_pwm(PWMPins.PWM1)
        print(f"Changing to frequency: {FREQUENCIES[counter % 4]} Hz")
        pwm.set_config(pwm_num=PWMPins.PWM1, frequency=FREQUENCIES[counter % 4])
        counter += 1
        time.sleep(2)
except Exception as exc:
    print(exc)
    print(f"Failed in {time.time() - start_time} seconds")
```

It will eventually fail. Sometimes it fails at the same time, sometimes it's at different times. The errors are almost always `FileNotFoundError`s or `PWMError`s.

sjpark608, 2023-05-31:
@jarell-santella @farzadpanahi 
Looks like some delay is required between reading/writing to the file.

- reading and writing of frequency are performed in two separate session
- one-second delay between reading and writing
- running for 3 minutes without error

jarell-santella, 2023-06-09:
This is actually how it is supposed to behave. Refer to the `__exit__` and `__del__` methods defined in the source code: https://python-periphery.readthedocs.io/en/latest/_modules/periphery/pwm.html#PWM. Both of these methods close the file that is responsible for PWM on the system.

In the second script, we have:
```py
while True:
    pwm = EdgePiPWM()
    pwm.init_pwm(PWMPins.PWM1)
```

On every loop, we are reassigning `pwm` to a new `EdgePiPWM` object. By doing this, we are making the previous `pwm` object eligible for garbage collection which then closes the file responsible for PWM on the system, which causes this issue.

closed 2023-06-09 by jarell-santella