EdgePi-Cloud/edgepi-python-sdk #377: RTD is noisy

farzadpanahi, 2023-08-02:
[image]
The RTD probe is just sitting on the desk in room temperature but we are seeing a wide range of change from 17 to 33 degrees C.

ikohOsensa, 2023-08-03:
Yea I am seeing the same issue on mine just measuring the resistance with an multimeter where I see a jump of anywhere from 2 ohms to 4 ohms every second or so. It feels like there is a mosfet or switch in there that is changing the junction circuit path sporadically causing the resistance to jump and thereby introducing the noise. Its a coin flip between software or hardware fix.

https://github.com/EdgePi-Cloud/edgepi-python-sdk/assets/92751554/7c2e3b75-2fca-4bd8-95df-ced67fb3e82a

farzadpanahi, 2023-08-03:
@CFraserOsensa if you get a chance, can you take a look at this ?

CFraserOsensa, 2023-08-03:
Each of the RTD leads go through an analog switch on their way to the ADS1263 chip. This adds resistance, but it basically just acts like wire resistance and should be steady. I know in the past Steven has tested bypassing the switches since we had our circuit wrong but I think in the last revision he was able to test this without issues to the extent you are seeing. 

There is a GPIO from the IO Expander called RTD_EN that we could probe to see if it looks steady
[image]

jarell-santella, 2023-08-05:
```py
adc.set_config(conversion_mode=ConvMode.CONTINUOUS, adc_2_data_rate=ADC2DataRate.SPS_800)
adc.set_rtd(True, adc_num=ADCNum.ADC_2)

while True:
    reading = adc.read_rtd_temperature()
    print(f"RTD read {reading} degrees celsius")
    time.sleep(1)
```

This will generate the same noise as well.

[image]

sjpark608, 2023-08-18:
[image]
Reading from my test setup.

- Only reading RTD and TC by controlling with SDK
- min 20 , max 28 observed over 60 seconds period
- This deviation is due to the filtering/noise, using slower datarate should reduce the noise range
- The resistance jumps that Ian observed seems like the change of configuration as the dashboard steps through different channels.

farzadpanahi, 2023-08-18:
@sjpark608 are you saying this is expected based on ADC datasheet? based on datasheet how much noise we should expect at highest data rate of adc2 for rtd? 
in your test if you use the slowest data rate, how much noise you will see?

sjpark608, 2023-08-19:
Page 29 of [datasheet](https://www.ti.com/lit/ds/symlink/ads1263.pdf?ts=1692329912908) shows the table of ENOB and Noise level according to the datarate. At 800SPS, ADC2 introduces about 3mV of peak-to-peak noise.

The ADC2 code value of reading RTD voltage directly proportional to the ratio of Vrtd and Vreference. Vreference can be calculated by multiplying Rref (2K ohms) by 2*IDAC(500uA) = 2V. Only one IDAC goes across the Vrtd, so Vrtd = Rrtd*500uA. This simplifies to ```code``` = ```Vrtd/(2*Vref)``` or ```code``` = ```Rrtd/(2*Rref)```. Therefore the Vnoise-p-p(3mV)/(2*Vref) * 2^23 = code = 6291 approx. ((6291)/2^23)*(2*Rref) = noise resistance = about 3 ohms the RTD conversion ratio is 0.385ohms/C. Noise tempeature is around 7.79 celcius.

sjpark608, 2023-08-22:
[image]
RTD has a very slow thermal response time. It really doesn't make sense to have a high data rate for RTD since it will just introduce adc internal noise to the measurement

farzadpanahi, 2023-09-14:
@sjpark608 could you do the noise calculation for 10S/s ? 
[image]
[image]

This is my edgepi and my rtd is running at 10S/s. the noise is still around a few degrees. I want to know if it is within spec.

sjpark608, 2023-09-14:
@farzadpanahi calculated noise peak to peak for 10sps is 0.07°C, but the RTD has error of +/- 0.5°C

farzadpanahi, 2023-09-14:
@sjpark608 so what we see is not within spec?

ikohOsensa, 2024-04-03:
[image]
Copying image over from other issue