EdgePi-Cloud/edgepi-python-sdk #378: ADC delay is not long enough

jarell-santella, 2023-08-05:
```py
adc.set_config(conversion_mode=ConvMode.CONTINUOUS, adc_2_data_rate=ADC2DataRate.SPS_800)
adc.set_rtd(True, adc_num=ADCNum.ADC_2)

while True:
    start_time = int(datetime.timestamp(datetime.now()) * 10**3)
    adc.set_rtd(True, adc_num=ADCNum.ADC_2)
    time.sleep(0.1)
    reading = adc.read_rtd_temperature()
    print(f"RTD read {reading} degrees celsius")
    adc.set_rtd(False)
    time.sleep(1)
```

That sleep in line 6 is necessary because the built-in sleep in SDK is not enough. If I remove that sleep between setting RTD and reading RTD, the reading will be -259.7402597402597 °C.

farzadpanahi, 2023-08-05:
@sjpark608  let's make sure we have integration tests for this

sjpark608, 2023-08-18:
@jarell-santella why are we enabling/disabling rtd in the loop?

farzadpanahi, 2023-08-18:
@sjpark608  I think this can technically happen if we have other channels set to use adc2.

sjpark608, 2023-08-18:
@jarell-santella 
@farzadpanahi 
in that case, you should use     ```reading = edgepi_adc.single_sample_rtd()``` this method instead. ```read_rtd_temperature``` uses a continuous reading delay which is shorter than the single sample delay.

[image]

I was able to read without sleep delay

farzadpanahi, 2023-08-18:
@jarell-santella I think the idea was to use cont. mode only if there are no other channels set to use adc2. in that case we turn on rtd and leave it on in cont. mode. we don't need to turn it on and off each time. 

as soon as a channel is configured to use adc2, then we should switch to single sample mode.