EdgePi-Cloud/edgepi-python-sdk #390: DAC setting to voltage 0 does not work

farzadpanahi, 2023-09-19:
To reproduce:
1. set out4 to digital high 
2. set out4 to analog 0v 
3. out4 shows it should be outputing 0v but the measurement shows 12v

Note A-OUT 4 in this image:
[image]

Logs from dev mgr:
```
 2023-09-19 13:16:01,412 DEBUG [update_handler_outputs.py:129] __read_outputs_deltas: module=DAC, deltas={'ch_4': {'voltage': 0}}
 2023-09-19 13:16:01,412 INFO [update_handler_factory.py:86] Factory is attempting to create update handler of DAC
 2023-09-19 13:16:01,413 INFO [update_handler.py:250] Parsing deltas for DAC. Deltas = {'ch_4': {'voltage': 0}}
 2023-09-19 13:16:01,414 DEBUG [update_handler_dac.py:204] gain not updated; not included in the delta
 2023-09-19 13:16:01,414 DEBUG [update_handler_dac.py:151] UpdateHandlerDAC: updating DAC channel voltage
 2023-09-19 13:16:01,414 DEBUG [update_handler_dac.py:175] Changing DAC channel DACChannel.AOUT4 voltage to 0 by calling self.dac.write_voltage(DACChannel.AOUT4, 0)
 2023-09-19 13:16:01,415 INFO [helper_funcs.py:30] Attempt 1: for <bound method EdgePiDAC.write_voltage of <edgepi.dac.edgepi_dac.EdgePiDAC object at 0x7f92b14220>> {'analog_out': <DACChannel.AOUT4: 3>, 'voltage': 0}
 2023-09-19 13:16:01,415 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,416 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [3], Msg Place Holder [255]
 2023-09-19 13:16:01,417 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [3],Msg Place Holder [67]
 2023-09-19 13:16:01,418 DEBUG [edgepi_gpio_expander.py:78] :read_expander_pin: pin 'DAC_GAIN' = 'True'
 2023-09-19 13:16:01,418 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,419 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [7], Msg Place Holder [255]
 2023-09-19 13:16:01,420 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [7],Msg Place Holder [2]
 2023-09-19 13:16:01,420 DEBUG [edgepi_gpio_expander.py:100] :get_expander_pin_direction: pin 'DAC_GAIN' = 'False'
 2023-09-19 13:16:01,421 DEBUG [dac_commands.py:37] Full code generated 0.0
 2023-09-19 13:16:01,421 DEBUG [dac_commands.py:57] Int code generated 0
 2023-09-19 13:16:01,422 DEBUG [edgepi_dac.py:124] Code: 0
 2023-09-19 13:16:01,422 DEBUG [spi.py:53] Open SPI device with path '/dev/spidev6.3'
 2023-09-19 13:16:01,423 DEBUG [dac_commands.py:132] Combined Command is: [51, 0, 0]
 2023-09-19 13:16:01,423 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,424 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [3], Msg Place Holder [255]
 2023-09-19 13:16:01,425 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [3],Msg Place Holder [160]
 2023-09-19 13:16:01,425 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,426 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [7], Msg Place Holder [255]
 2023-09-19 13:16:01,427 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [7],Msg Place Holder [0]
 2023-09-19 13:16:01,427 DEBUG [edgepi_gpio_expander.py:100] :get_expander_pin_direction: pin 'AO_EN4' = 'False'
 2023-09-19 13:16:01,428 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,428 DEBUG [edgepi_gpio_expander.py:58] Write Message Content <periphery.i2c.I2C.Message object at 0x7f90ffecd0>
 2023-09-19 13:16:01,430 DEBUG [edgepi_gpio_expander.py:184] __apply_code_to_register: dev_addx='32', reg_addx='0x3', reg_val='0xa0, opcode='OpCode(op_code=0, reg_address=3, op_mask=191)'
 2023-09-19 13:16:01,430 DEBUG [edgepi_gpio_expander.py:211] :clear_expander_pin: pin 'AO_EN4' = set to low
 2023-09-19 13:16:01,430 DEBUG [dac_commands.py:132] Combined Command is: [147, 0, 0]
 2023-09-19 13:16:01,431 DEBUG [spi.py:53] Open SPI device with path '/dev/spidev6.3'
 2023-09-19 13:16:01,432 DEBUG [edgepi_dac.py:181] reading code [147, 0, 0]
 2023-09-19 13:16:01,433 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,434 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [3], Msg Place Holder [255]
 2023-09-19 13:16:01,435 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [3],Msg Place Holder [67]
 2023-09-19 13:16:01,435 DEBUG [edgepi_gpio_expander.py:78] :read_expander_pin: pin 'DAC_GAIN' = 'True'
 2023-09-19 13:16:01,436 DEBUG [i2c.py:34] Open I2C device with path '/dev/i2c-10'
 2023-09-19 13:16:01,436 DEBUG [edgepi_gpio_expander.py:37] Read Message: Register Address [7], Msg Place Holder [255]
 2023-09-19 13:16:01,437 DEBUG [edgepi_gpio_expander.py:40] Message Read: Register Address [7],Msg Place Holder [2]
 2023-09-19 13:16:01,438 DEBUG [edgepi_gpio_expander.py:100] :get_expander_pin_direction: pin 'DAC_GAIN' = 'False'
 2023-09-19 13:16:01,438 DEBUG [dac_commands.py:109] code_to_voltage: code 0x0 = 0.0 V
 2023-09-19 13:16:01,438 INFO [helper_funcs.py:34] <bound method EdgePiDAC.write_voltage of <edgepi.dac.edgepi_dac.EdgePiDAC object at 0x7f92b14220>> {'analog_out': <DACChannel.AOUT4: 3>, 'voltage': 0} ran successfully. Returning True
```
from these logs it looks like that the voltage zero is properly set.

sjpark608, 2023-09-20:
@farzadpanahi 
https://github.com/EdgePi-Cloud/edgepi-python-sdk/blob/c9b3a8d47e5dcfa9741e1ebe92da0cfc34cd86c1/src/edgepi/dac/edgepi_dac.py#L79-L92

the condition on 87, it only disables the Analog Outpin. This is why you'd see 12V when 0 is written to the DAC

farzadpanahi, 2023-09-20:
and why do we have that condition?

sjpark608, 2023-09-20:
> and why do we have that condition?

due to the limitation of the circuit, DAC cannot produce true 0V. So we wanted to disable the Analog Out to set it to Digital ground. There should've been another line clearing the corresponding digital output state.

farzadpanahi, 2023-09-20:
so this can be fixed : ) ?

sjpark608, 2023-09-20:
> so this can be fixed : ) ?

yes, I'm on it :)

sjpark608, 2023-09-21:
Cases to think about
DAC:
1. Ch3~8:
   1-1. DAC is written with voltage 
   1-2. DAC is written with 0V
   1-3. Digital Out was High and DAC is written with voltage ```possible contention```
   1-4. Digital Out was Low and DAC is written with voltage 
   1-5. Digital Out was High and DAC is written with 0V ```possible contention```
   1-6. Digital Out was Low and DAC is written with 0V
Solutions:
1-3. AOUTx enabled is first, then DOUTx is  switched OFF, this way the LOWER side FET is never activated
1-5. AOUTx is enabled first, then DOUTx is switched OFF, DAC gets written 0V, then disable AOUTx,

2. Ch1/2
   2-1. DAC is written with voltage 
   2-2. DAC is written with 0V
   2-3. Digital Out was High and DAC is written with voltage ```possible contention```
   2-4. Digital Out was Low and DAC is written with voltage 
   2-5. Digital Out was High and DAC is written with 0V ```possible contention```
   2-6. Digital Out was Low and DAC is written with 0V
   2-7. PWM was enabled and DAC is written with voltage ```PWM can be seen for short time```
   2-8. PWM was enabled and DAC is written with 0V ```PWM can be seen for short time```
Solutions:
2-3. AOUTx enabled is first, then DOUTx is  switched OFF, this way the LOWER side FET is never activated
2-5. AOUTx is enabled first, then DOUTx is switched OFF, DAC gets written 0V, then disable AOUTx,
2-7. PWM_EN pin is disabled first, AOUTx is enabled
2-8. PWM_EN pin is disabled first, write 0 to DAC

DOUT:
1. Ch3~8:
   1-1. DOUTx is written HIGH
   1-2. DOUTx is written LOW
   1-3. AOUTx was at some V and DOUTx is written HIGH  
   1-4. AOUTx was at 0V and DOUTx is written HIGH ```possible contention```
   1-5. AOUTx was at some V and DOUTx is written LOW
   1-6. AOUTx was at 0V and DOUTx is written LOW
Solutions:
1-4. AOUTx is enabled, DOUTx is enabled, this will disengage the low side FET, then disable AOUTx

2. Ch1/2:
   2-1. DOUTx is written HIGH
   2-2. DOUTx is written LOW
   2-3. AOUTx was at some V and DOUTx is written HIGH  
   2-4. AOUTx was at 0V and DOUTx is written HIGH ```possible contention```
   2-5. AOUTx was at some V and DOUTx is written LOW 
   2-6. AOUTx was at 0V and DOUTx is written LOW
   2-7. PWM was enabled and DOUTx is written HIGH
   2-8. PWM was enabled and DOUTx is written LOW

PWM:
1. PWM1/2
   1-1. DOUT1/2 HIGH -> PWM1/2 Enable
   1-2. DOUT1/2 LOW -> PWM1/2 Enable
   1-3. AOUT1/2 X volt -> PWM1/2 Enable
   1-4. AOUT1/2 0 volt -> PWM1/2 Enable
   1-5. DOUT1/2 HIGH -> PWM1/2 Disable
   1-6. DOUT1/2 LOW -> PWM1/2 Disable
   1-7. AOUT1/2 X volt -> PWM1/2 Disable
   1-8. AOUT1/2 0 volt -> PWM1/2 Disable